vdudouyt/minipro #31: Suggested change of project description

DavidGriffith, 2015-01-09:
It seems more aesthetically pleasing to have the following text as the project's description: "An Open Source program for controlling the MiniPRO TL866xx series of chip programmers.".  It should also help casual wanderers find software for burning chips under Linux.

vdudouyt, 2015-01-09:
Hi,

Thank you very much (I'm not a native English speaker, so I realize that my sentences may sound unnaturally). I changed it in the way you suggest.

Best Regards,

Valentin

DavidGriffith, 2015-01-09:
Looks good.  Thanks!  I'm so very glad to have found your program.  A lot of people have been clamoring for something like this for a long time.  I would have written one myself, but I had no idea where to begin.

closed 2015-01-09 by DavidGriffith