vdudouyt/minipro #33: Downloadable release source archive (with a version number)

ranauei, 2015-01-26:
Hello I'd like to add this utility to MacPorts and it gets complicated without a versioned source archive (I can't use the generic master zip file that github provides I need one with a version number in it). It'll be really nice if you could provide one.

vdudouyt, 2015-01-27:
Hi!

I uploaded it as a .tar.gz archive into https://github.com/vdudouyt/minipro/releases. Is it now enough to be packaged in MacPorts?

Best Regards,

Valentin

ranauei, 2015-01-27:
Sorry but the file I'm getting is a 5mb blob that seems to be a cat of all the files in the repo :(

ranauei, 2015-01-27:
Thanks I created and submitted the new MacPort

closed 2015-01-28 by vdudouyt

ranauei, 2015-02-13:
Minipro is now available on MacPorts (committed today).