vdudouyt/minipro #34: Unknown device

AriesT, 2015-02-03:
System UBUNTU 14.04 x64 (TL866CS (minipro))

Have this:
./minipro -p am29f040 -r am29f040.bin
Unknown device

Thanks

vdudouyt, 2015-02-03:
Hi,

Please consider trying it in the following way:
```nohighlight
./minipro '-p 'AM29F002B @DIP32' -r am29f040.bin
```

Best Regards,
Valentin

AriesT, 2015-02-04:
Have this:

/minipro# ./minipro '-p 'AM29F002B @DIP32' -r am29f040.bin
>

 (cursor is active line as terminal with Ctrl+C as quit)

kutis96, 2015-04-18:
The device name must be inside single quotes, like this: ./minipro -p 'AM29F040B @PLCC32' -r am29f040.bin

richud, 2015-08-31:
in Ubuntu you need sudo , e.g.
```sh
$ sudo ./minipro -p "M27C256B @DIP28" -r testread.bin
[sudo] password for xxxxxx: 
Found Minipro TL866A v3.63.2
Chip ID OK: 0x208d
Reading Code... OK
```

As otherwise you get this;
open("/dev/bus/usb/004/007", O_RDWR)    = -1 EACCES (Permission denied)

Valentin, could you kindly update the README.md and program options menu (-p)  to be more explicit that you need both the chip name AND package type suffix?

closed 2019-02-27 by DavidGriffith