vdudouyt/minipro #44: Some ATtiny fuse support missing

gleepy, 2015-07-14:
Most newer ATtiny devices use protocol ID 0x73. No case in the fuse select code added fuse support. I have a code patch to apply to main.c to (hopefully) fix this issue, but it has only been partly tested with the ATtiny13A.

scotty264, 2017-01-14:
Hello:
I know this is an old issue but I do not see the code patch to support the particular TINY devices listed.  I can test the tiny25/45/85 and tiny84.

Thank you.

Peace and blessings,
Scott

LightningStalker, 2017-09-02:
Nothing happens when attempting to read the fuses on an ATtiny85.

LightningStalker, 2017-09-02:
Just putting a case for protocol 0x73, I get the following:
```
$ ./minipro -p attiny85 -r b.bin -c config
Found Minipro TL866A v03.2.80
Chip ID OK: 0x1e930b
Reading fuses... IO error: expected 15 bytes but 7 bytes transferred
```

LightningStalker, 2017-09-04:
A little port sniffing reveals that command 10 is being sent instead of command 12. I'm not sure what is causing this.

LightningStalker, 2017-09-05:
I was a break line left out in my programming. I'll have a pull req soon. I can barely flash an LED.

LightningStalker, 2017-09-05:
The MiniPro does not send the fuse extended byte on the ATtiny85 through command `0x12` as it does with the ATmega chips. This warrants further investibubblygation.

LightningStalker, 2017-09-05:
Pull req submitted

scotty264, 2017-09-05:
LightningStalker:
I have Tiny84's and Tiny13's I can test code on.  Just let me know.

Peace and blessings,
Scott

LightningStalker, 2017-09-05:
Ok cool, the modified code is in [the pull request](https://github.com/vdudouyt/minipro/pull/84).

LightningStalker, 2017-09-05:
I added [a new branch with Tiny84 enabled](https://github.com/LightningStalker/minipro/tree/tiny_fuses) if you want to test it.

elosha, 2018-02-15:
Today I tested the ATtiny45 – reading and programming low/high fuse via the socket worked like a charm.

The config file created omitted the extended fuse byte though. I just checked the datasheet and the byte is available on the Tiny45.
`fuses_lo = 0x00e1
fuses_hi = 0x005f
lock_byte = 0x00ff`
("fuses_ext" should be there)

scotty264, 2018-02-16:
Lightningstalker: I didn't see the patch for the Tiny84 until "elosha" posted yesterday and I received an email about it.  Sorry for the delay. 

I tested the ATtiny84.  Good read for code, eeprom and fuses but as "elosha" stated, the extended fuse is missing as you noted in your posts.  

fuses_lo = 0x0062
fuses_hi = 0x00df
lock_byte = 0x00ff

I also checked the ATtiny13 and it tested good for code, eeprom and fuse reads. The ATtiny13 only has high, low and lock fuses, no extended fuse.

Just wondering how the extended fuse is handled for parts like the ATmega328p. I have not looked at the source code but why is there an issue on the ATtiny25/45/85 and ATtiny84 reading the extended fuse?  

Peace and blessings

elosha, 2018-02-16:
The reason seems to be that there are multiple levels of fuse support. Implemented in fuses.c are, among others, `avr2_fuses` (upper, lower and lock) and `avr3_fuses` (upper, lower, extended and lock).

In devices.h, ATtiny13 is classified as `variant = 0x10`, but ATtiny45, ATtiny85 etc. too. ATtiny84 is has three fuses and `variant = 0x12`.
The `variant` byte is evaluated in main.c:551..555. In case of variant `0x10` and `0x12`, `avr2_fuses` gets assigned.

But I don't know whether it is wrong to classify as variant `0x10` or to assign `avr2_fuses` to it. Because if ATtiny13 has no extended fuse, the current setting works for it. I don't know what to actually check for in order to correctly assign `avr3_fuses`.

ATmega328P is correctly assigned avr3_fuses in main.c:542 :)

elosha, 2018-02-16:
@LightningStalker, do you need help by someone checking implementation of the whole ATtiny series (or AVRs) for fuse layout compliance?

scotty264, 2018-02-16:
Eliza:
I looked at the source code back in Sept. 2017.  I recall what you are stating and was also confused about how to properly code for the tiny84 and tiny25/45/85. Though, I haven't had the time to sit down and figure it out.  Thanks for your effort. :)

Peace and blessings.

rozhuk-im, 2018-03-09:
You can play with different chips params in mine fork: https://github.com/rozhuk-im/minipro
create your test chips DB file and change every think that you want.
If you need - I can move fuses to external ini files too.

LightningStalker, 2018-03-10:
It traced back to lines 159 and 177 of `minipro.c`. The Minipro expects `0x03` for parts the use protocol ID 0x73 and the correct number of bytes.

scotty264, 2018-03-10:
rozhuk-im: your build requires cmake v3.1. I'm at v2.8.9. Changing line 6 in the file "CMakeLists.txt" to "CMAKE_MINIMUM_REQUIRED(VERSION 2.8 FATAL_ERROR)" allowed cmake to run. However, make fails with: 

`minipro/src/main.c: In function ‘print_usage’:
minipro/src/main.c:253:2: warning: passing argument 1 of ‘__xpg_basename’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from minipro/src/main.c:13:0:
/usr/include/libgen.h:35:14: note: expected ‘char *’ but argument is of type ‘const char *’
minipro/src/main.c: At top level:
minipro/src/main.c:265:26: error: expected ‘;’, ‘,’ or ‘)’ before ‘__unused’
minipro/src/main.c: In function ‘main’:
minipro/src/main.c:449:36: error: ‘progress_cb’ undeclared (first use in this function)
minipro/src/main.c:449:36: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [src/CMakeFiles/minipro.dir/main.c.o] Error 1
make[1]: *** [src/CMakeFiles/minipro.dir/all] Error 2
make: *** [all] Error 2`

Any ideas how to correct? 

I am using "gcc version 4.7.3 (Ubuntu/Linaro 4.7.3-2ubuntu1~12.04)". I'm not a very proficient "C" programmer as assembly is my game. :)

Thank you for your efforts on this minipro interface.

Peace and blessings. Scott

scotty264, 2018-03-10:
In main.c, commented out line 13 [libgen.h], which cleared the error on line 253. On line 265, there was a space between " mp __unused", so corrected it to " mp__unused". That cleared the errors on line 449. Make compiles but fails at the linking stage with:

`make clean && make
[ 16%] Building C object src/CMakeFiles/minipro.dir/main.c.o
[ 33%] Building C object src/CMakeFiles/minipro.dir/minipro.c.o
[ 50%] Building C object src/CMakeFiles/minipro.dir/database.c.o
[ 66%] Building C object src/CMakeFiles/minipro.dir/liblcb/src/utils/ini.c.o
[ 83%] Building C object src/CMakeFiles/minipro.dir/liblcb/src/utils/sys.c.o
/media/truecrypt1/Projects/Active/TL866CS_Programmer/Software/minipro/src/liblcb/src/utils/sys.c: In function ‘write_pid’:
/media/truecrypt1/Projects/Active/TL866CS_Programmer/Software/minipro/src/liblcb/src/utils/sys.c:101:7: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
[100%] Building C object src/CMakeFiles/minipro.dir/liblcb/src/utils/buf_str.c.o
Linking C executable minipro
gcc: error: missing argument to ‘-l’
make[2]: *** [src/minipro] Error 1
make[1]: *** [src/CMakeFiles/minipro.dir/all] Error 2
make: *** [all] Error 2
`
I haven't yet figured out where the error is created. Perhaps this is a cmake v2.8.9 vs. cmake 3.1 difference?

rozhuk-im, 2018-03-10:
#ifndef __unused
#	define	__unused		__attribute__((__unused__))
#endif
it defined in utils/macro.h, I forget to include this.
gcc have no __unused.
But renaming/delete "__unused" is ok as workaround/fix.

I check build on:
Linux ubuntux64 4.4.0-57-generic #78-Ubuntu SMP Fri Dec 9 23:50:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
it is ok, after few changes.

I unable to reproduce link error, probably because using newer cmake and gcc.

scotty264, 2018-03-10:
Hi rozhuk-im:
I'm confused... I deleted "__unused" at line 266. Still complaints about "__xpg_basename" from libgen.h, so I commented out libgen.h.

Still linking error. Any suggestions on what to look for on the linking error?

rozhuk-im, 2018-03-11:
basename() - generates only warning during compilation.
Mine build log on ubuntu:
```
[ 14%] Building C object src/CMakeFiles/minipro.dir/main.c.o
/root/minipro/src/main.c: In function ‘print_usage’:
/root/minipro/src/main.c:254:34: warning: passing argument 1 of ‘__xpg_basename’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  fprintf(stderr, usage, basename(progname));
                                  ^
In file included from /root/minipro/src/main.c:13:0:
/usr/include/libgen.h:34:14: note: expected ‘char *’ but argument is of type ‘const char *’
 extern char *__xpg_basename (char *__path) __THROW;
              ^
[ 28%] Building C object src/CMakeFiles/minipro.dir/minipro.c.o
[ 42%] Building C object src/CMakeFiles/minipro.dir/database.c.o
[ 57%] Building C object src/CMakeFiles/minipro.dir/liblcb/src/utils/ini.c.o
[ 71%] Building C object src/CMakeFiles/minipro.dir/liblcb/src/utils/sys.c.o
[ 85%] Building C object src/CMakeFiles/minipro.dir/liblcb/src/utils/buf_str.c.o
[100%] Linking C executable minipro
[100%] Built target minipro
```
Can you update cmake and gcc?

elosha, 2018-03-11:
@rozhuk-im, thank you for your reference to your hard work. What changes did you do to solve the problem? Could you two please discuss build problems of a different fork elsewhere? No offense, it just gets off-topic otherwise.

We already located the problem, we only need a _logical rule_ now to know if a device has two or three fuses just by looking at the programmer's config/identification bytes. This is more like a knowledge problem. Then it seems be easy to build a working if-condition for newer ATtiny devices.

@LightningStalker To me this place looks obscure. Wouldn't matching selection of `avr*_fuses` hand over the correct `length` parameter to this function automatically?

rozhuk-im, 2018-03-11:
@elosha +1 off-topic.
I had never work with ATtiny devices, my use cases very simple.
Is fuses described in datasheets?

elosha, 2018-03-11:
@rozhuk-im: Yes, they are, it really is simple to see if they also have an 'extended' fuse or just 'high' and 'low'.

Sadly, the `variant` byte in devices.h seems to be in use by the programmer hardware (minipro.c:47), so we can't just change the variant bytes and assign `avr3_fuses` to the variant value (main.c:551 ff), is that correct?

rozhuk-im, 2018-03-11:
device->variant is something like internal proto selector, I don't know what it mean, but it db constant.
HINT: 0x10063 - is really 0x0063, 0x10000 is some flag that not used.

If you describe/show where to read how to select proper fuses then I can try to implement it.
Or you could just add to BD ini file (in fork) ATtiny devices with different fuses and select what you need then run minipro.

LightningStalker, 2018-03-11:
@elosha it does in most cases but protocol 0x73 behaves differently from the others.

elosha, 2018-03-11:
Ok, I finally got it! All devices with `protocol_id = 0x73` and ( `variant = 0x10` or `variant = 0x12` ) have three fuses (`avr3_fuses` instead of `avr2_fuses` in main.c:554). Excecpt the ATtiny13/13A/13V devices, which have two fuses only. 

How to handle this exception? Either by id, they all share `0x1e9007`, or by`code_memory_size = 0x400` (or less). I prefer the latter since there can be more/future devices that are missing a fuse because of their reduced size, but have a different id.

```
		case 0x73:
		  switch(device->variant) {
			case 0x10:
			case 0x12:
				if(device->code_memory_size  <= 0x400) // small devices of this series only have 2 fuses
					device->fuses = avr2_fuses;
				else
					device->fuses = avr3_fuses;			
				break;
		  }
		  break;
```

They also have a distinct opts1/2/3/4 value, maybe there's a useful flag hidden in there.

I also find the flag method interesting, that @rozhuk-im pointed out: Add flag with variant in devices.h, apply special processing, remove flag from variant before sending it to the programmer.

LightningStalker, 2018-03-11:
The problem with msg[2] in minipro.c still needs to be addressed.

scotty264, 2018-03-12:
elosha: 
I'm a little confused about your statement " Either by id, they all share 0x1e9007".  What parts are "they all" referring to? In scanning devices.h, the signature bytes of "0x1e9007" only belong to the ATtiny13x series of parts. The signature bytes are always different for the different "base" part numbers but can be the same for the sub-parts, i.e. ATtiny13, ATtiny13A and ATtiny13V.

I did successfully implement your code change and it works for the ATtiny13 series but fails for ATtiny11 and ATtiny12 devices as a fuse structure is ever defined for "variant = 0x00", only "variant = 0x10" or "variant = 0x12".

The follow code snippets adds back support for ATtiny11/12 and uses the ATtiny13x signature bytes to select the proper fuse structure for the ATtiny13x series. I have succesffully tested the fuse read/write code with ATtiny11x, ATtiny12x and ATtiny13x:

`	case 0x73:
		switch (device->variant) {
		case 0x00:		// ATtiny11 and ATtiny12
			device->fuses = avr_fuses;
			break;
		case 0x10:
		case 0x12:
			if (device->chip_id == 0x1e9007)	// small devices of this series only have 2 fuses (ATtiny13x)
				device->fuses = avr2_fuses;
			else
				device->fuses = avr3_fuses;
			break;
		}
		break;
`

The original code does work with the ATtiny11/12.

I'm also not functional with GIT, so anyone capable of adding this code snippet the the source tree, please do so.

elosha, 2018-03-13:
Dear @scotty264, thanks for bringing the ATtiny11/12 situation to attention, very important!

> What parts are "they all" referring to? In scanning devices.h, the signature bytes of "0x1e9007" only belong to the ATtiny13x series of parts.

Correct, with "they all" I meant the ATtiny13x sub-models. But I did't really like to incorporate specific device information into the fuse support logic, hence the idea with detecting pruned protocol-0x73-variant-0x12 devices by code size.

You even tested it – does the code @LightningStalker pointed to (minipro.c:159/177) still need care or does it actually work?

In the future, we really should include a device's fuse information into the upcoming external device DB. Only with this approach minipro can offer device definition without changing program code. Before ATtiny13x, fuses looked much more systematic. 😆

scotty264, 2018-03-14:
elosha:
I just tested writing fuses to ATtiny11, ATtiny12 and ATtiny13 and they programmed just fine. What is the issue with msg[2]?

LightningStalker, 2018-03-14:
Do you have an ATtiny84 and 85 to test? Those are the chips I was running into problems with IIRC.

DavidGriffith, 2018-03-14:
I have some ATtiny85.  Exactly what should I do to test?

LightningStalker, 2018-03-14:
Read and write fuses

chenz, 2018-03-14:
Last time, (#84) the issue was that the ATTINY85 part (proto 0x73, variant 0x10) simply did not work with 
`avr3_fuses` because of the read size issue - I don't think anything has changed in this regard since then?

DavidGriffith, 2018-03-14:
@LightningStalker I'm looking for something we can put in this Issue to say "type this exactly to verify this fix"...

scotty264, 2018-03-14:
All: 
I have ATtiny84A.  Tried with fuse read and MINIPRO errored using "minipro -p "ATTINY84A" -c config  -r config.txt && echo ; cat config.txt" and no joy. 

Error encountered.
**Found Minipro TL866A v03.2.80
Chip ID OK: 0x1e930c
Reading fuses... IO error: expected 10 bytes but 9 bytes transferred****

elosha, 2018-03-14:
Just to eliminate an option, does minipro *supply the correct protocol and variant to the programmer*? The Windows tool was fixed once in this regard, V5.91 changelog from 2013-03-09:

> Fix：ATTINY24/44/84/25/45/85 Fuse Bits Program

Does devices.h contain buggy info from before the fix or is it up-to-date?

scotty264, 2018-03-14:
elosha:
As I found out, ATtiny84/85 share the same protocol and both fail in fuse read/write. 

But how would I know if devices.h contains buggy info or not? What can I use to compare it with?

LightningStalker, 2018-03-15:
@scotty264 try changing line 147 of minipro.c to `msg[2] = 0x03;`
I don't recall weather I was able to get writing to work. That happens on line 157. We need to change the tests on those lines to send `0x03` for these ATtinies without breaking functionality for the other chips.

scotty264, 2018-03-17:
LightningStalker:
I changed line 147, 159 and 177 to `msg[2] = 3;`.  After doing so, both reading and writing of fuses for ATtiny84/85 worked correctly.

I see there is already a conditional on all three of those lines for use with PICs.

LightningStalker, 2018-03-17:
Yes we need a conditional that works for all devices.

DavidGriffith, 2018-03-20:
Could I see a branch where all this tinkering is going on?

I just now emailed @vdudouyt about where devices.h came from.

elosha, 2018-03-20:
Could someone help me to understand minipro.c:159?
`msg[2] = (length==4)?0x02:0x01;  // 2 fuse PICs have len=8`
If `length` equals 4, there are two fuses, else there is only one. Ok.
Then the comment claims that two-fuse PICs have `length` 8, this seems contradictory?

I'm trying to understand that part in order to come up with an universal way to calculate values of msg[2] (length).

scotty264, 2018-03-20:
@elosha:
The conditional operators `?` and `:` are the short form of the if-else statement. It is small but effective. Sometimes also known as "ternary operator" because it takes three arguments.  I have read that some compilers produce more code for this variation (over "if-else" statements) on certain MCU's.

Maybe the comment is simply in error and should read `// 2 fuse PICS have len=4"`.

Line 159 is the same as line 147 and line 177 just expressed using HEX based numbers. I don't see any reason why "if - else if -else" statements cannot be used for the ATtiny-specific case.

Peace and blessings.

DavidGriffith, 2018-03-21:
I strongly suggest that people resist any temptation to use the ternary operator.

elosha, 2018-03-21:
Thank you, @scotty264. The ternary operator wasn't my problem (except for its name 😄), just the comment saying something else than the condition. Either two-fuse PICs have len=4 or the ternary operator got mixed up and was meant to read `msg[2] = (length==4)?0x01:0x02;` but I don't know which is true now …? Yes, let's get rid of it.

scotty264, 2018-03-21:
@elosha: I didn't think you didn't know that operator but the wording was confusing, so I thought it best to explain it just in case. From what I've read elsewhere, both the ternary operator and the if-else statements compile to the same code on the AVR with avr-gcc.

chenz, 2018-03-22:
@elosha `pic2_fuses` has *both* length 4 (2 config words) and 8 (4 user words). These hacks are meant to detect when the config words of a `pic2_fuses` device are transferred, and only in that particular case `msg[2]=2` is set.

For the AVR devices I tried, i cannot make sense of the parameters - then again, they *do* use different protocols, so different rulesets may be at play.

ATTINY2313A (proto 0x71, variant 0x6b)
- three fuse device, correctly set up as to use avr3_fuses in main.c 
- happily reads three fuses with `msg[2]=1`

ATTINY85 (proto 0x73, variant 0x10)
- three fuse device, "incorrectly" set up as to use avr2_fuses in main.c
- happily reads two fuses with `msg[2]=1`
- read error when patched to use avr3_fuses
- reads ok when patched to use avr3_fuses *and* `msg[2]=3`

Interestingly for a lot of AVR devices, I found a corelation between the actual number of fuses and the `write_unlock` member of device_t. It is unclear where these values came from, since the don't appear in the "official" device database (see #109).

closed 2019-02-27 by DavidGriffith