vdudouyt/minipro #48: AM28F020A support?

alexforencich, 2015-09-01:
I have a whole tube of these things, but it seems that only the AM28F020 is supported.  The AM28F020A has a built-in sequencer that does most of the work, simplifying the erase and program process.  However, this makes it completely incompatible with the AM28F020 mode.  Since it is a much simpler protocol, it should not be very difficult to implement.

elosha, 2018-03-11:
You could try to use a EEPROM device protocol. Just make sure it uses the same programming voltage as your device and ignore id mismatch. If it works, I suggest someone could duplicate that device and replace the id.

DavidGriffith, 2018-03-14:
@alexforencich Are you still here?  What package are your chips in?  I'd like to get one of those chips to play with.

closed 2019-02-27 by DavidGriffith