vdudouyt/minipro #49: Doesn't seem to work on OS X El Capitan

samw3, 2015-11-14:
I've installed libusb using brew.  I then build with the Makefile.osx, and everything seems to go fine.  I then plug in my TL866CS programmer from autoelectric.cn with a W27C512 int the 40 pin and run this command line. 

./minipro -p "W27C512 @DIP28" -r "test.hex"

Nothing is printed and the command never returns.  However, if I unplug the programmer, I get:

IO error: bulk_transfer: LIBUSB_ERROR_IO

Any ideas?

samw3, 2015-11-15:
I did some debugging,  it hangs on:
    msg_recv(handle, buf, 40);
in minipro_get_system_info

samw3, 2015-11-15:
Further:

	ret = libusb_bulk_transfer(handle->usb_handle, (1 | direction), buf, length, &bytes_transferred, 0);

inside msg_transfer is the point it hangs, though it doesn't previously go through that line.  Though never as a read.

peternoyes, 2015-12-29:
I can reproduce this issue. I am trying:
minipro -p "AT28C256 @SOIC28" -w test.bin

The RUN light blinks once, but then nothing happens.

samw3, 2015-12-30:
@peternoyes It appears to be an Apple bug.  The libusb team submitted a bug report to Apple but we haven't seen a fix yet.  See here: https://github.com/libusb/libusb/issues/123

d235j, 2016-01-27:
This appears to be fixed with in 10.11.4 Beta 2.

samw3, 2016-03-22:
It is indeed fixed in 10.11.4

closed 2016-03-22 by samw3