vdudouyt/minipro #51: Version number mismatch compared to Windows app

Malvineous, 2016-02-03:
I just started using my new TL866A and I had to upgrade the firmware, however while this program worked fine, I noticed that it is reporting a different version number to the Windows application:

Before the reflash the Windows program reported device firmware 3.2.58, while this application reported 3.58.2.  After the reflash the Windows program told me I was now on firmware version 3.2.72, while this app is reporting 3.72.2.

Hopefully it's just a simple matter of swapping the version numbers so it's consistent with the Windows app!

tomsim, 2016-04-28:
Fix all ready exist from 9 Jan : https://github.com/vdudouyt/minipro/pull/50 
I do not know why it is not merged yet?

vdudouyt, 2016-04-28:
@tomsim sorry, I merged it now.

tomsim, 2016-04-28:
@vdudouyt You can close this issue

closed 2016-04-28 by Malvineous