vdudouyt/minipro #58: Can't program gal 22v10d with jedec file

daybyter, 2016-08-31:
Hi!

I'm trying to burn a jedec file into a Lattice 22v10d GAL:

=============
andreas@Elektronik2:~/pal_gal/c64_z80/gal$ miniprohex -p GAL22V10D -w 
z80logic.jed
Found Minipro TL866CS v03.2.72
Incorrect file size: 1764 (needed 5892)
andreas@Elektronik2:~/pal_gal/c64_z80/gal$
=============

Nu success so far. What am I doing wrong?

Thanks in advance,
Andreas

kutis96, 2017-04-12:
This error tends to happen whenever you're supplying some invalid sort of data into the thing.
It seems the programmer actually needs the data in _some weird bitstream_ the MiniPro's micro can just spit into the GAL/PAL/whathaveyou, instead of the straight JEDEC file. If you _really_ need a GAL/PAL in your design, you either are:

- ever so slightly screwed
- destined to figure out a way to convert the JEDEC file into the bitstream the MiniPro/GAL can understand. This may actually be quite helpful of you to do. 😄 

The JEDEC format is pretty well-documented, and it's from a long time ago, so it should be easy enough to parse [_citation needed_]. 

I may look into this later today, but don't be too mad at me if I don't. 😃

rlake, 2017-10-17:
Ah! I was optimistic there would be GAL support but couldn't see how it would work from the source. It's a shame the MiniPro protocol is proprietary, but I can at least confirm the JEDEC format is simple to decode. I ran this through a translator and wrote some python to parse and build a fuse array with no trouble: [http://elm-chan.org/docs/dev/jed.html](http://elm-chan.org/docs/dev/jed.html)

These fuses then need to get to the device somehow as per [GAL Programming Algorithms](http://www.armory.com/~rstevew/Public/Pgmrs/GAL/_ClikMe1st.htm) and the (critically unknown) MiniPro abstraction. I wonder, does the MiniPro require a chip present to start programming? It may not actually matter as presumably the first action is to read the PES block so it's feasible somebody with one of these and a logic analyser to RA* and serial pins could figure it out. But yeah, long shot...

DavidGriffith, 2018-03-13:
Possibly related to #96.

Cyb3rn0id, 2018-10-09:
Someone has solved this issue? I've the Minipro V6.82, the TL866A V03.2.85 and a Lattice GAL22V10B. After programming gives verify error at address 32

DavidGriffith, 2018-10-10:
All development has moved to Gitlab.  See https://gitlab.com/DavidGriffith/minipro

closed 2018-10-10 by DavidGriffith