vdudouyt/minipro #61: NAND

anselmru, 2016-11-21:
Приветствую! Мне сказали, что minipro не поддерживает программирование NAND-флеш через адаптер TSOP48. Так ли это? Нельзя ли добавить в таком случае новый протокол ".protocol_id" для программирования NAND? Или это принципиально невозможно? Или для этого нужно менять прошивку самого программатора? Или подскажите мне пожалуйста, где я мог бы найти информацию, чтобы самому добавить такой протокол? Благодарю за труд!
-----------------
Hello! I was told that the minipro does not support the programming of NAND flash using adapter TSOP48. Is it really so? Is it possible to add in this case a new Protocol ".protocol_id" for programming NAND? Or is it fundamentally impossible? Or do we need to change the firmware of the programmer? Or tell me please, where could I find information to add such a Protocol? Thank you for your work!

vdudouyt, 2016-11-21:
Hello,

I'll reply you in English as not all Github users understand Russian.

No, it is not possible to add a new device without modifying firmware unless it's supported by proprietary tool.

closed 2016-11-21 by vdudouyt

anselmru, 2016-11-21:
Thank you for the clarification!