vdudouyt/minipro #63: Firmware is too old / Core dumped

jefm, 2016-11-24:
Hello minipro fans,
I'm new to PLD and trying all this out. 
Ubuntu 16.04.1 LTS, minipro 0.1, TL866A
from lsusb: Bus 002 Device 042: ID 04d8:e11c Microchip Technology, Inc. TL866CS EEPROM Programmer [MiniPRO]

Here's what I get.

$ sudo minipro -p "ATF16V8B" -r testread.bin
Warning: firmware is too old
Found Minipro TL866A v03.2.58
Floating point exception (core dumped)

The little Minipro lamp blinks when this happens so something is reaching the hardware at least.
On core dump, I get the Ubuntu "splat" screen. Sorry, getting info out of it is vague. I've updated everything and my system whines about some Gnome conflict. I'd imagine that doesn't hurt minipro, but I've no idea. 
Thanks for any help.

jefm, 2016-11-24:
Just for kicks, I took the chip out (ATF16V8B) and the same error occurs.

Sentinel77, 2016-11-25:
I would update your Minipro firmware to the current version v03.2.72
You can do this by downloading a newer version of the Minipro Windows software.
You can also do the firmware update on Linux with TL866_Updater https://github.com/radiomanV/TL866

DavidGriffith, 2018-03-13:
Firmware warning not related to floating point exception.  See #96.

closed 2018-03-13 by DavidGriffith