vdudouyt/minipro #70: Windows Support?

ryansturmer, 2017-02-09:
Any idea what it would take to support windows?  This seems like a great tool, and the need for a command line access to the MiniPro under window is great.

Palmitoxico, 2017-02-15:
Maybe is possible to build under MSYS/Cygwin ?

zacaj, 2017-09-21:
Hey, just stumbled on this project when looking for the same thing, wanted to automate rom burning in my build script.  I took a quick stab at this and was able to get the program to compile under mingw-64 fairly easily, and successfully dumped a rom with it.  Haven't played with it too much, as part of the requirements for windows libusb requires replacing the minipro usb driver with a generic one they provide, which makes the minipro GUI not work, so every time you want to use the other one you have to reinstall the drivers...  

If someone wants to actually turn this into a pull request I can upload my changes, but they're a bit minimal and hacky right now as I'm not super familiar with Makefiles and unix C development...  (or even making pull requests :/ )

DavidGriffith, 2018-02-14:
@zacaj If you can fork this repo and make a branch in your account with these changes, I'll be glad to go over your work.

timofonic, 2018-03-10:
@zacaj Yes please!

closed 2019-02-27 by DavidGriffith