vdudouyt/minipro #85: Using make on OS X 10.13

Emulty, 2017-10-14:
Hi there!
I can't compile minipro on macOS v10.13 & Xcode 9.0

`
OS:minipro User$ make
make: pkg-config: Command not found
cc -g -O0    -c -o byte_utils.o byte_utils.c
make: pkg-config: Command not found
cc -g -O0    -c -o database.o database.c
make: pkg-config: Command not found
cc -g -O0    -c -o minipro.o minipro.c
make: pkg-config: Command not found
cc -g -O0    -c -o fuses.o fuses.c
make: pkg-config: Command not found
cc -g -O0    -c -o easyconfig.o easyconfig.c
make: pkg-config: Command not found
cc -g -O0    -c -o main.o main.c
make: pkg-config: Command not found
cc -g -O0    -c -o minipro-query-db.o minipro-query-db.c
make: pkg-config: Command not found
cc byte_utils.o database.o minipro.o fuses.o easyconfig.o main.o  -o minipro
Undefined symbols for architecture x86_64:
  "_libusb_bulk_transfer", referenced from:
      _msg_transfer in minipro.o
  "_libusb_claim_interface", referenced from:
      _msg_transfer in minipro.o
  "_libusb_close", referenced from:
      _minipro_close in minipro.o
  "_libusb_error_name", referenced from:
      _minipro_open in minipro.o
      _msg_transfer in minipro.o
  "_libusb_init", referenced from:
      _minipro_open in minipro.o
  "_libusb_open_device_with_vid_pid", referenced from:
      _minipro_open in minipro.o
  "_libusb_release_interface", referenced from:
      _msg_transfer in minipro.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [minipro] Error 1
OS:minipro User$
`
Could you help me?

chenz, 2017-10-14:
Try `make -f Makefile.osx`

Emulty, 2017-10-15:
Many thanks! It solved my problem.

nmaupu, 2018-01-02:
In case it helps someone, I had an issue using Mac OS.
I Installed libusb using brew :
```
brew install libusb
brew link libusb
```

But when compiling, libusb.h was not found by the compiler...

To fix, I changed the Makefile to the following:
```
CFLAGS	?= -g -O0 -Wall
```
(notice the ? added before = sign)

Then I compiled with :
```
CFLAGS="-I /usr/local/include/libusb-1.0 -g -O0 -Wall" make -f Makefile.osx
```

chenz, 2018-01-02:
All Makefile variables [can already be overridden](https://www.gnu.org/software/make/manual/html_node/Overriding.html#Overriding) by default like this:

    make -f Makefile.osx CFLAGS="..."

Note that "make" does the overriding here, no environment variable involved.

nmaupu, 2018-01-02:
Oh! Nice I didn't know that... It indeed works as expected without changing anything. Thanks for the tip !

DavidGriffith, 2018-02-14:
Can this issue be safely closed?

DavidGriffith, 2018-03-04:
Does Pull #94 fix this problem?

DavidGriffith, 2018-03-04:
Fixed by #94

closed 2018-03-04 by DavidGriffith