vdudouyt/minipro #87: Using the TSOP-48 adapter fails, but works in Windows

skumlos, 2018-01-08:
If I use the official TSOP-48 adapter (original V3), I seem to read garbage from the chip (A MX29LV160T). It reads extremely fast, much faster than the official software. Now if I try to read it in Windows first, and then in Linux, it seems to work... Programming, reading everything... Reading then goes slower.

Any clues?

It's almost as if some variable is undefined by this software, which is set by the Windows software, and apparently survives being software disconnected from a virtual Windows machine to the Linux host.

alfishe, 2018-01-09:
Seems adapter has it's own microcontroller that needs to be initialized from minipro. after it's done - adapter works just fine

skumlos, 2018-01-09:
Yeah so the original software most likely sends a command to the unit, when the adapter is needed (or something like that). I guess the only way to find out is to sniff some data...

raybellis, 2018-05-16:
I also need TSOP support, and can't get it working.  I'm trying to program an AT29C040A (TSOP32) with what as far as I know is an official V3 TSOP48 adapter.

closed 2019-02-27 by DavidGriffith