vdudouyt/minipro #89: ST M27C256B chips all fail strangely

DavidGriffith, 2018-02-04:
I have a tube of ST-branded M27C256B UV-erasable EPROMs that all fail to have stuff written from 0x200 to 0x27F.  The first bad byte written is 0x00.  Subsequent bad bytes are 0xFF.  Then from 0x280 on, the chip contains the expected data.  Would this be caused by a failure in the chips or might there be something wrong with the controlling program?  I don't have any other means of programming chips.

DavidGriffith, 2018-02-21:
Does anyone here have a working programmer from the 1980s or 1990s and a UV eraser?  I'd like to provide you with one or two of these chips so we can determine what the problem is.  It could be trouble in this software, trouble in the tl866's firmware, or simply a bad batch of chips.

This chip is still popular enough that I feel it's important to get to the bottom of the problem.

szg0000, 2018-02-25:
I have UV EPROMs, I will try to write with minipro.

DavidGriffith, 2018-02-28:
That exact brand and part number?

elosha, 2018-03-11:
I don't have an eraser, but if you could send one device of your batch to Germany, I could try to program using a Willem or TaskIt. For the sake of advance in this issue ;)

DavidGriffith, 2018-03-12:
Could you email me with your physical address?

DavidGriffith, 2018-03-21:
I sent CPUWIZ over at Atari Age two of these chips which he programmed on a Xeltek 610P without trouble.

@elosha, To get to the bottom of this, I think you'll need to get an eraser.  This problem appears to be in the TL866 CS/A firmware or in devices.h.

See http://atariage.com/forums/topic/275512-m27c256b-eproms-and-minipro-tl866-trouble/?p=3989251 for CPUWIZ's test.

DavidGriffith, 2018-03-21:
I found a data sheet for this part: http://ecee.colorado.edu/~mcclurel/STMicro_M27C256B_EPROM_2384.pdf

Programming voltage is 12.75V +/- 0.25V

elosha, 2018-03-21:
Thank you, David. I'm trying to buy an eraser.

Really wonder why this exact device is causing trouble. Maybe the EPROM chips didn't turn out 100% perfect, the TL866 programs slightly out of specification and the combination of both makes the error appear. But just a guess …

#107 could be helpful in this issue. For example, I could check for correct programming voltage more easily if I knew how it is defined.

DavidGriffith, 2018-03-21:
Please hold on buying that eraser for a bit.  I have a couple more tests I'd like to run involving the official software and different firmware versions.

DavidGriffith, 2018-03-28:
I just checked programming some M27C256B chips with our minipro program and the official program.  I tried with firmware versions 3.2.61, 3.2.80, and 3.2.81.  Each time with our minipro, the programming failed exactly as I described above.  Each time with the official program (versions 5.91, 6.60, and 6.70), the programming succeeded without a hitch.  So, I'm positive now that the problem is with our implementation.

DavidGriffith, 2018-03-28:
Word of warning... don't try to flash your tl866 using @radiomanV's program while you have Virtualbox running.  At least for me, Virtualbox will automatically latch onto the tl866.  When the connect and disconnect happens with the firmware updater, Virtualbox will grab the tl866 and cause a failed erase or write.  Fortunately this is easily recovered from by just trying again, or shutting off Virtualbox.

elosha, 2018-03-28:
Thanks for the hint, be careful! 🔨  I have access to an old Windows XP laptop for updating.

So I will get an UV eraser. The Chinese are in that business, too, with a cheap and tiny device for around 15€ on Ebay. Is it usable?

DavidGriffith, 2018-03-28:
I have no experience with those Chinese erasers.  The one I use is something made 30+ years ago in California.

DavidGriffith, 2018-03-28:
I forgot to mention that I used @radiomanV's firmware updater compiled for Linux, not the Windows version.  I sent him a pull-request that updated the README.md file to include instructions on how to compile it for Linux.

DOSputin, 2018-04-28:
Had similar problems with EPROM M27C256B that were branded as `ST M27C256B Singapore` and date coded between 1000-1140.  

> _The first bad byte written is 0x00. Subsequent bad bytes are 0xFF._

Yeah,  that sounds familiar. 

By chance I discovered that I could read and write these UV chips properly by using the definition `SMJ27C256 @DIP28` instead of the recommended `M27C256B @DIP28`   
Another large batch of the `ST M27C256B Korea` have no problems with the I/O using the standard definition.

DavidGriffith, 2018-06-02:
@elosha, I have not done so.  Is the physical address you sent me earlier still valid?  Shall I send you a chip now?

elosha, 2018-06-03:
@DavidGriffith, I was just wondering if it has been lost. As your questions have been answered, I don't know if I can find out anything new to help fix the issue. But if you want to give it a try and have too many chips, the address is still valid. ;)

shea1337, 2018-06-13:
I'm having the same issues with a TL866CS and some TI and AMD branded 27c040. Seems like all these chips want 12.75v +/- 0.25v though so hopefully 12.50 gives it the right amount of power.

closed 2019-02-27 by DavidGriffith