vdudouyt/minipro #90: RFE/RFD: Defining your own chip configs

valgrind, 2018-02-05:
Developers that only occasionally dump stuff for Mame often have a TL866, which works well enough for EPROMs, but the database does not cover many PROMs. The PROMs are usually very simple; address lines, data lines, GND, VCC, and chip enable. Current method of dumping them is usually to pull out an arduino, breadboard and a handful of wires. Then rewire for each chip type. On the small board I'm working on now there are 3 diffrent chip types.

It would be nice if we could just add the chip definition to the TL866 database and recompile instead. So is there an ELI5 for what to write in the database to get a specific pinout just for reading? Writing is not interesting.

For reference, here is the data cheet for one of the ones I'm about to dump: http://hxc2001.free.fr/Squale/datasheet/TBP18S030.pdf

DavidGriffith, 2018-02-14:
This is a rather tricky thing to do because the host computer doesn't do that much to read or write chips.  All that stuff is handled by the firmware and Autoelectric holds the source code.  If a new chip is to be supported, it needs to be added to the firmware.  This program doesn't do much more than tell the burner what chip to expect and what to write to it.  

That being said, I've heard some poorly-corroborated stories that the TL866 can be made to arbitrarily bitbang various pins, which might allow you to do what you propose.

DavidGriffith, 2018-02-15:
I think it would be better to dig into @radiomanV's work to find the answer.

pbortas, 2018-02-15:
It would definitely be better, question is if I'll ever get around to it. Well, it's on my future projects list now, closing this.

closed 2018-02-15 by pbortas