vdudouyt/minipro #92: Last proper release 0.1 is quite old now

elosha, 2018-02-15:
First, thanks to the maintainer and contributors for providing this important tool for embedded development on open systems.

I use the TL866 on Mac OS X and ran into a weird problem today, when I noticed that it would not read the config file of my ATtiny45. I checked the issues and found a showstopper that made reading of fuses on ATtiny impossible – but it had already been fixed long ago. It took me some moments to realize that Mac Ports does rely on the 0.1 release, which is four years old now … and quite buggy.

May I suggest to tag a new minor release like 0.1.1 in the near future, so packagers can notice that and finally catch up with some important fixes. ;)

rozhuk-im, 2018-03-09:
Try more fresh fork: https://github.com/rozhuk-im/minipro :)

DavidGriffith, 2018-03-10:
@rozhuk-im, I'm having trouble getting your fork compiled.  Would you please prepare a pull request incorporating your changes minus the cmake stuff and file movements?  We can work on those things later.

rozhuk-im, 2018-03-10:
I was make pull req year ago, before migrate to cmake, liblcb, removing files and other big/hard changes, but get no response from @vdudouyt here and by email. I wait for ~7 month.
Now there is to many changes, and I dont want drop it.
CMake do some tests and set some defines, that used in minipro and liblcb.
Some files removed because they not needed any more, some  because I do not know what they do or cant test that they work as intended.

About compilation, try now, if it fail - welcome to new issue req :)

DavidGriffith, 2018-03-10:
I looked at your changes and I agree, there are a LOT of changes.  Could we begin the process of porting them to @vdudouyt's repo?  The first commit after the point of divergence (b03c273f4172543da2b6c9e617dd6266b1c20733)  is quite extensive.  Could you help me with that commit (https://github.com/vdudouyt/minipro/commit/cebc708694ec76c3ec6f95c525db880e068972d9) to understand just what was changed?  I'd like to break it down into individual commits per issue fixed/enhanced.

rozhuk-im, 2018-03-10:
https://github.com/vdudouyt/minipro/commit/cebc708694ec76c3ec6f95c525db880e068972d9
* most changes - code style/formatting changes.
* change int and char to other types in many places
* add dev_db_dump_flt() and add call for it in few places
* make all communications with chips db via internal api (remove extern device_t devices[])

About bug fixes - I do not remember now, but:
* add few memset to ensure that no crap from stack/mem reuse in buffer before send command/write
* in msg_init() strange assign: buf[8] = buf[6]; commented out
* use local buf on stack to send commands (in future replaced by special allocated buf) instead of some global buf

May be more easy way is to revert last commits, stash something, and merge?
Or some black magic like:
```
cd /tmp/
git clone git@github.com:vdudouyt/minipro.git
cd minipro
git remote add upstream git@github.com:rozhuk-im/minipro.git
git fetch --all
git checkout --track -b master_rim upstream/master
git checkout master_rim
git merge -s ours master
git checkout master
git merge master_rim
git branch -D master_rim
git submodule init
git submodule update --remote
git push
git push origin master
rm -rf /tmp/minipro
```

elosha, 2018-03-11:
I wonder what this has to do with my request to properly release a new version number.

Packagers like Mac Ports don't take the master branch for their packages but look out for the latest stable version, even if its many years old. It's irrelevant how many changes were made afterwards in the master branch and if many bugs were fixed, because it counts as unstable. So the solution is a new official bug fix _release_ that reflects the changes of the _past_.

DavidGriffith, 2018-03-11:
@elosha Sorry about that.  I started a new issue to talk about this merge at Issue #98.

@rozhuk-im, would you please copy your comments about your fork to Issue #98?

Now, about the new version number, I suggest we use 1.0.0 and follow what is laid down at https://semver.org/.  Comments?

elosha, 2018-03-11:
Thanks, David :)

I suggest going from major 0.x to 1.x when some significant problems are ironed out. But to my feelings are that we are pretty close.

What I think is "significant" are those serious issues that actually prevent use of minipro with several popular devices:
- [ ] #44 
- [ ] #96 
… as well as finding out if these are device errors:
- [ ] #89 
- [ ] #82 

I also think those bugs should have priority over adding cool features like an external device database.

DavidGriffith, 2018-03-12:
@vdudouyt has explicitly authorized me to direct things with minipro.  I'll go with the single decimal point scheme as you suggest.  Those four bugs certainly are showstoppers.  Let's focus on solving those whereupon we can tag and announce 1.0.

DavidGriffith, 2018-03-25:
While we're waiting for things to happen for 1.0, would anyone be opposed to tagging a 0.2 release?

closed 2019-02-27 by DavidGriffith