vdudouyt/minipro #96: Floating point exception error

szg0000, 2018-03-06:
I you give `minipro -p "GAL16V8" -r readfile.jed` minipro returnes with Floating point exception error.

DavidGriffith, 2018-03-06:
What's the maker and part number of the chip?  Can you post an example jed file?

szg0000, 2018-03-06:
The chip doesn't  matter. Just type this command with or without the chip.
Can you reproduce this bug? I am on P4, 32 bit system.

szg0000, 2018-03-06:
Same problem with x86_64.

DavidGriffith, 2018-03-07:
I've confirmed this bug.

```
$ minipro -p "GAL16V8" -r foo.jed
Found Minipro TL866A v03.2.80
Floating point exception
$
```

DavidGriffith, 2018-03-07:
The FPE occurs at line 194 in main.c in read_page_ram().

```
/* RAM-centric IO operations */
void read_page_ram(minipro_handle_t *handle, unsigned char *buf, unsigned int type, const char *name, int size) {
        char status_msg[24];
        sprintf(status_msg, "Reading %s... ", name);

        device_t *device = handle->device;
        int blocks_count = size / device->read_buffer_size;  /* FPE happens here */
        if(size % device->read_buffer_size != 0) {
                blocks_count++;
        }
```

DavidGriffith, 2018-03-07:
The FPE happens because in ```devices.h```,  ```read_buffer_size``` for GAL16V8 is 0x00.  This is also true for several other devices, all GALs.  Similarly, these devices also have 0x00 for ```write_buffer_size```.  Attempting to write to a GAL will also result in a FPE because of line 224 in main.c in ```write_page_ram()```.

```
void write_page_ram(minipro_handle_t *handle, unsigned char *buf, unsigned int $
        char status_msg[24];
        sprintf(status_msg, "Writing %s... ", name);

        device_t *device = handle->device;

        int blocks_count = size / device->write_buffer_size;
        if(size % device->write_buffer_size != 0) {
                blocks_count++;
        }
```

DavidGriffith, 2018-03-07:
Would someone who better knows the low-level details of writing a GAL please comment on this?  Why are the read and write buffer sizes set to zero?

szg0000, 2018-03-07:
Analyze the communication between the Minipro and the Windows [software](http://www.autoelectric.cn/MiniPro/minipro_setup670.rar) with [wireshark](https://www.wireshark.org/).

rozhuk-im, 2018-03-11:
IMHO, if db contain write_buffer_size=0 or read_buffer_size=0 then, may be we should try to read/write whole size that user request.

closed 2019-02-27 by DavidGriffith