vdudouyt/minipro #97: udev files put in the wrong place for local compilations

DavidGriffith, 2018-03-10:
This change: 52786b1c1bf893751b0ed25f5f1405a6bb0be239 changes the location where minipro's udev file is written to from ```/etc/udev/rules.d``` to what renders as ```/lib/udev/rules.d```.  This is okay for binary packages (ie debs), but not when minipro is compiled and installed from source.  Some mechanism to detect if a package is being built and to use ```/etc/udev/rules.d``` if this is not the case.

DavidGriffith, 2018-03-10:
It looks like Pull #68 was trying to solve some sort of problem that was partially solved by (and muddled up by) 52786b1c1bf893751b0ed25f5f1405a6bb0be239.  We'll have to keep #68 in mind when addressing this issue.

chenz, 2018-03-10:
My change from #81 simply fixed the hard coded paths used before. Personally, I don't  see any problem with installing under /lib if that is the location that pkg-config tells you. But clearly, there are still other issues with the whole approach, such as the installation as non-root.

#83 was an attempt to fix these, but the changes in there quickly became too complicated and are still fragile. 

I would now propose to instead make the installation explicit - ie have separate targets for (un-) installing the rules, make them independent from "make install" but document them in README . Also document the variables that can be overridden to customize the install location (or simply direct advanced users to study the Makefile).

I can create a new PR if there is consensus on how to proceed.

#68 is mostly orthogonal to this issue.

DavidGriffith, 2018-03-10:
I've always been brought up to never put stuff into /lib/ or /usr/lib/ when installing locally-built packages -- that I should leave writing there to the initial installation of the OS or the package management.

I'll have to study #83 when my mind is fresher.

What do you, @chenz, propose be done with #68?

chenz, 2018-03-10:
Installing into /etc is worse imo ;-)

#83 was a maybe misguided attempt to have "make install" automatically do the right thing. The idea was to mimic other packages that have udev rules and bash-completion files, but those packages typically use a much more involved build process than a simple Makefile, and things got a bit too complex. 

That is why I would now prefer to just drop the installation (bash-completion as well) from Makefile and leave it to the (self-compiling) user (ie document it) and the distro packaging.

#68 looked okay to me, I just suggested renaming the file instead of splitting it up because it seemed more elegant to me. I am not an expert on udev though, and I don't know if there is a safe/portable way to determine the correct priority values. If this is too distro-specific (it may well be because of the group names used may not be standardized either), it should be implemented in the Debian packaging.

chenz, 2018-03-12:
I have now implemented my proposed changes in #100. Regarding #68 - i have looked at related Debian packages (avrdude, dfu-util), and they use the default priority of 60, so that should work for minipro as well (I have adjusted the Debian packaging accordingly).

closed 2018-03-14 by DavidGriffith