vdudouyt/minipro #98: Merge in changes from https://github.com/rozhuk-im/minipro

DavidGriffith, 2018-03-11:
A year and a few months ago, @rozhuk-im made extensive changes in his own fork and @vdudouyt never replied to his pull request.  As time rolled on, the two forks diverged to such a degree that a clean merge is no longer possible.  @rozuk-im's changes are very useful and I feel we should begin the process of manually applying his changes to @vdudouyt's repository.

DavidGriffith, 2018-03-11:
Copied from #92:

@rozhuk-im, I looked at your changes and I agree, there are a LOT of changes; too many to simply make a new pull request.  Could we begin the process of porting them to @vdudouyt's repo?  The first commit after the point of divergence (b03c273f4172543da2b6c9e617dd6266b1c20733) is quite extensive.  Could you help me with that commit (https://github.com/vdudouyt/minipro/commit/cebc708694ec76c3ec6f95c525db880e068972d9) to understand just what was changed and why?  I'd like to break it down into individual commits per issue fixed/enhanced.

I can now compile your fork.  Thanks.

elosha, 2018-03-11:
Changes by @rozhuk-im look great, but I have mixed feelings about cmake (goal of portability, opposed to use of 'fresh‘ features or tools) and nomenclature changes (professional term 'device' -> 'chip', which rather is slang). But maybe there are good reasons for this.

rozhuk-im, 2018-03-11:
Copied form: https://github.com/vdudouyt/minipro/issues/92#issuecomment-372062995

https://github.com/vdudouyt/minipro/commit/cebc708694ec76c3ec6f95c525db880e068972d9
* most changes - code style/formatting changes.
* change int and char to other types in many places
* add dev_db_dump_flt() and add call for it in few places
* make all communications with chips db via internal api (remove extern device_t devices[])

About bug fixes - I do not remember now, but:
* add few memset to ensure that no crap from stack/mem reuse in buffer before send command/write
* in msg_init() strange assign: buf[8] = buf[6]; commented out
* use local buf on stack to send commands (in future replaced by special allocated buf) instead of some global buf

May be more easy way is to revert last commits, stash something, and merge?
Or some black magic like:
```
cd /tmp/
git clone git@github.com:vdudouyt/minipro.git
cd minipro
git remote add upstream git@github.com:rozhuk-im/minipro.git
git fetch --all
git checkout --track -b master_rim upstream/master
git checkout master_rim
git merge -s ours master
git checkout master
git merge master_rim
git branch -D master_rim
git submodule init
git submodule update --remote
git push
git push origin master
rm -rf /tmp/minipro
```

rozhuk-im, 2018-03-11:
@elosha 
There was "device" for MiniPRO TL866xx and "device" for chip.
CMake is more easy to use than makefiles and autotools, at least for me. One file for project and one file per binary. CMake do many things without my interaction.

elosha, 2018-03-11:
@rozhuk-im: I see. The project could use an naming convention to make terms distinguishable.

DavidGriffith, 2018-03-14:
Perhaps call the tl866 a "programmer"?

elosha, 2018-03-14:
Maybe it is even necessary to write model specific code and programmer universal code in the near future. There are now three different programmers with different device lists … just something that came to my mind.

closed 2019-02-27 by DavidGriffith