vdudouyt/minipro #101: New Programmer: TL866II Plus

elosha, 2018-03-14:
XGecu alias Autoelectric has just released the [TL866II Plus](http://www.autoelectric.cn/EN/TL866_MAIN.html) with new features.

Main changes:
- minimum Vcc 1.8V support (was 3.3V minimum)
- NAND flash support
- maximum Vpp reduced to 18V (was 21V maximum)
- 512MB flash support (was 128MB maximum)
- Pin detection –> ?

TL866A/CS are no longer in production.

Regarding the device list, some ICs will be supported by the TL866II Plus only (especially Vcc 1.8V devices), some by TL866A/CS only (Vpp 21V devices) and some by TL866II Plus and TL866A only (those needing the ICSP port).

DavidGriffith, 2018-03-15:
The reduction of maximum Vpp has me concerned.  I'd love to see their justification for that.

elosha, 2018-03-20:
Guess it is a question of voltage range and they moved it down as whole to reach 1.8V.

There was an official 1.8V device adapter for TL866CS/A, but I don't know if minipro supports that and got these devices covered.

ee-muhammad, 2018-05-21:
My observation is that they offer separate windows PC application. I fear that this project may be incompatible with new version of programmer. Has anyone tested it?

DavidGriffith, 2018-05-22:
@ee-muhammad to date, none of the main developers, myself included, have obtained a TL866 II+ yet.  I presume it's different enough from the CS/A variants such that simply running minipro as-is won't work.  A big stumbling block for getting support for this new variant going is finishing up on extracting the device database from the official controlling software.

PeterBortas, 2018-05-22:
I have one I ordered for future possible work with proghq. Not planning to spend a lot of time with it until proghq's support for TL866 is solid, but if someone wants some quick tests run I can probably find some time for that.

AJPoulter-Soton, 2018-05-26:
Just FYI I do have the new TL866 II Plus (by mistake, I actually ordered the CS model :-( but this is what was delivered). I can confirm that it *doesn't* work with the software as is (everything results in an "Error connecting to device" error). I suspect this is due to a differed USB device ID? (I've never really done much with USB - so a long way from being an expert). If I get time later on, I'll try to put it though a debugger and see what (if anything) can be done if I can spot how to fix that.

AJPoulter-Soton, 2018-05-26:
So I was sort of right, the libusb_open_device_with_vid_pid seems to fail; but changing the vid / pid to the values for the new II+ move things on; but then we get 'IO error: bulk_transfer'. Not sure if this info  has helped anyone. I'll see if I can spend a bit more time looking at this in a week or two.

AJPoulter-Soton, 2018-06-01:
After having done a bit more digging, it seems that the TL866 II Plus isn't compatible with the TL866 CS / A (http://proghq.org/wiki/TL866#TL866_II_PLUS_is_NOT_compatible) – so it will likely be quite a lot of work to make it usable with this software.

I suggest (at least in the short-term) that an error message stating that the software isn't compatible with the newer hardware might at least help a user understand the problem… I'll see if I get chance to put something together as a PR to do that, over the weekend.

DavidGriffith, 2018-06-02:
@AJPoulter-Soton I agree with this approach.  Though, I suspect that the differences between the two products are not particularly different.  I think that the II+ is initialized differently and the command structure is slightly different to account for more capabilities.

Auctoris, 2018-06-03:
PR raised to report incompatibility if the TL866 II+ is detected.

AJPoulter-Soton, 2018-06-03:
In a "Sunday afternoon & not quite paying attention" kind of way, I note that I've raised the PR using my personal (as opposed to academic) account (@Auctoris)... So apologies for any confusion caused.

DavidGriffith, 2018-06-03:
@AJPoulter-Soton I suppose it was best that your personal email address was the one that went into the modification history.

hcsaustrup, 2018-09-26:
+1 for compatibility with TL866 II+ :-)

DavidGriffith, 2018-10-10:
All development has moved to Gitlab. See https://gitlab.com/DavidGriffith/minipro

closed 2019-02-27 by DavidGriffith