vdudouyt/minipro #107: Few documentation about TL866 and comments in code

elosha, 2018-03-21:
As we a trying to add and fix support for some devices, as well as the new programmer model, it becomes apparent that documentation about

1. how the TL866 communicates (e.g. which part is firmware controlled)
2. how protocol messages are put together (e.g. lengths, fuses)
3. what the values in devices.h represent (e.g. package detail, variant, opts)

is scarce.

May I suggest to start a documentation project to aid code contributors? It really would help contributors to understand the programmers how things work.

DavidGriffith, 2018-03-21:
Has anyone been able to get in touch with anyone at Autoelectric?

PeterBortas, 2018-03-21:
> I spent several hours last night tracking down as many of the known releases of the official software as I could

Nice. Could you host them somewhere publicly,or send them to me and I'll do that?

DavidGriffith, 2018-03-21:
Still remaining to be found are versions 5.80, 5.70, and 1.00.

chenz, 2018-03-23:
As for the style of documentation, my preference is having it side-by-side with the code in the editor,  so either text or markdown files in the repository (or even *in* the code - [doxygen](http://www.stack.nl/~dimitri/doxygen/) has markdown support these days). 

That make it easier to keep documentation in sync with the code, especially when working with branches.

DavidGriffith, 2018-03-25:
@chenz: Can you come up with a sample repo of how this would look?

closed 2019-02-27 by DavidGriffith