vdudouyt/minipro #115: int size appears in a few places

DavidGriffith, 2018-04-19:
I found some places where a size for something is defined as an int.  We should probably be using size_t instead.

rozhuk-im, 2018-08-17:
Bad idea use size_t every where, int was very bad.
size_t designed to address host memory size, but this struct device contain external flash memory size.
For example, off_t is used to address some disk/storage offset/size, and sizeof(off_t) not depend on x32/x64 cpu arch.
TL866 uses 24 bit address, so uint32_t - is OK for addresses and flash size (in db - struct device).
get_file_size() - off_t!!!
compare_memory() - size_t - OK
other size_t - ok, it is for in app mem calculations.

DavidGriffith, 2018-08-19:
Are there any other places where I shouldn't be using size_t besides the type for get_file_size()?

rozhuk-im, 2018-08-19:
`struct device` and all other places where "size" - mean size/offset for flash

closed 2019-02-27 by DavidGriffith