vdudouyt/minipro #122: With firmware versions after 6.71 mini pro doesn't work.

firew4lker, 2018-07-27:
With firmware versions after 6.71 mini pro doesn't work. It exits wuth error:

IO error: bulk_transfer: LIBUSB_ERROR_OVERFLOW

c32ff0, 2018-07-28:
same . 
i had it working on a raspberry pi but i wanted to program a gal22v10 and used a windows pc which required me to reflash. the same day, on the raspberry pi it would hang at random points during the upload and the next day it whouldn't do anything, just hang.
on my laptop i get the IO error: bulk_transfer: LIBUSB_ERROR_OVERFLOW error

DavidGriffith, 2018-08-08:
Were you able to get Minipro to program a gal22v10 at all under an earlier firmware? We have an ongoing problem with programming GAL devices. See #58.

Please followup at https://gitlab.com/DavidGriffith/minipro/issues/121.

DavidGriffith, 2018-08-10:
A solution has been found.  See it on Gitlab.

rozhuk-im, 2018-08-17:
Or use mine: https://github.com/rozhuk-im/minipro
It have no problem even without any code patch, because it have read buf 4096 bytes (instead 40 bytes in unpatched original and 44 for fixed version on gitlab).
And will not have any same class problem in future.

closed 2019-02-27 by DavidGriffith