vdudouyt/minipro #123: Winbond W27C512 64k EEPROM Device ID Mismatch

DOSputin, 2018-08-01:
Seems that there is some error or device id offset in the device database?

`$ minipro -p "W27C512 @DIP28" -w data.bin
Found Minipro TL866CS v03.2.81
Invalid Chip ID: expected 0xda08, got 0xda02`

DOSputin, 2018-08-01:
The EEPROM i'm using are labeled 
>Winbond W27C512-45z

Tried random EEPROM device settings until one was found that reported the expected HEX `0xda08`
This device type was  `ICE27C512`
`minipro -p "ICE27C512 @DIP28" -w data.bin'  will work if a known blank `W27C512` is used. 
However, if the device is not empty, the erase voltage is not high enough to clear the memory and the write verification fails.

c32ff0, 2018-08-02:
i think this repo is supposed to be abandoned
"The current repo is at https://gitlab.com/DavidGriffith/minipro"

DOSputin, 2018-08-08:
moved & expanded issue here: https://gitlab.com/DavidGriffith/minipro/issues/122

rozhuk-im, 2018-08-18:
You can use fork: https://github.com/rozhuk-im/minipro
Edit or create your own chip database - minipro_db.ini file.
No recompilation required.

PS:
chip_id=0xda02:
W27C257 @DIP28, W27C257 @PLCC32, W27E257 @DIP28, W27E257 @PLCC32

closed 2019-02-27 by DavidGriffith